3drobotics/solodevguide #322: ETA for programmatic video pipeline

ayushajain, 2016-07-22:
The current video pipeline through vidlaunch seems to be causing constant errors and timeouts. Additionally, using the libcamsolo library does not give us complete control over the camera resolution and advanced gopro filters. Any idea when the new video api is going to be released?

On the solo developer guide, there doesn't seem to be a release date for the api.
[https://dev.3dr.com/concept-video.html](url)

mrpollo, 2016-07-25:
@ayushajain sadly there's no ETA on a public video api, what are you trying to do?, perhaps its easier for us to help you on your specific project rather than try to get the full public API out.

alexblack, 2016-07-25:
I'm also interested!  Something I'd like to be able to do is to use `libsolocam` to capture higher resolutions than 1280x720, changing frame rates would be good too

franknitty69, 2016-07-26:
i just need access to the camera for cv (object recognition, object avoidance, landing on a moving platform).

ayushajain, 2016-07-26:
I'm also interested in an updated libsolocam. We're trying to use the drone to detect smaller objects from about 200 feet, and we need to be able to access a higher resolution to do this.

mrpollo, 2016-07-29:
hey guys here's an update given that this topic has gathered enough attention, 3DR has no plans, I repeat no plans to support a video pipeline API, I'm sorry guys but you are on your own, the Solo Forums are a great place to find Solo resources like the following: https://discuss.dronekit.io/t/solo-video-acquire-please/320/25

proficnc, 2016-07-30:
Any chance that the files to make an image for both imx 6's could just be released? 

It's not like there is any reason to keep it secret now....

Please?

GPSFan, 2016-08-03:
I'd also like to request that 3DR (or what's left of it) release the kernel sources to build the exact kernel version that is being shipped with the solo.
With the code available here:
http://solo-packages.s3-website-us-east-1.amazonaws.com/
and here:
https://3dr.com/about/open-source/
it is possible to rebuild a working kernel and kernel modules that will allow the solo's userland to function properly.
However there must have been some changes since that code was released because the kernel version reported by a running solo is:
3.10.17-rt12-1.0.0_ga+g4176d57
Not:
3.10.17-rt12-1.0.0_ga+g3f15a11
which is what can be built from the supplied sources.