3drobotics/solodevguide #325: Connecting directly to the USB port using Accessory Breakout board

lovettchris, 2016-09-20:
The hardware-accessorybay.html page says "The IMX6 will connect in device mode if the 3DRBUS pin is not connected to GND. In device mode you can connect the Solo to a computer."

But I can't get this to work, my PC says "Windows has stopped this device because it has reported problems. (Code 43).  A request for the USB device descriptor failed."

Can you add more information on how to get iMX6 board to operate in device mode so I can connect the Solo to a computer?

hamishwillee, 2016-09-20:
@mrpollo Is there anyone at 3DR who could comment on this? That information was all that was provided when I created this, but I no longer have access to anyone who might be able to offer more.

mrpollo, 2016-09-20:
I have no idea if theres anyone who can answer that question, I personally never used this feature, sorry.

@liamstask perhaps you can answer this?

liamstask, 2016-09-20:
@lovettchris how do you have your board wired up at the moment? the `3DRBUS` pin referenced on that page does not exist - I just edited that to refer instead to the `3DRID` pin. With an accessory bay breakout board, you should just need to plug into the USB connector without any other modifications in order for Solo to connect as a USB device.

As a debug step, you could ground the `3DRID` pin to configure Solo as a USB host and then connect a test device (flash drive or similar) to verify that your USB hardware is working in general. It's possible that there could be an issue with the cable internally, for instance.

cc @kevinmehall

franknitty69, 2016-09-21:
i've never been able to get solo in device mode to do anything. as a matter of fact, because the usb 5v pin is still hot in device mode, you can destroy your computer. that's what happened to my macbook pro when i connected it to a solo in device mode via the accessory bay board usb.

even in device mode connected to other devices such as to an intel edison in otg mode, the solo does nothing.

i've only gotten the accessory bay to work in when in host mode. in host mode it works as described.

clovett, 2016-09-30:
Agreed, in fact, I think I've also damaged the USB on the drone by plugging it into my computer.  So someone should update the 3dr solo documentation to tell folks more clearly not to do that.

alvikag, 2018-03-05:
Hi,
I am trying to get real time sensor data from solo over mavros. Can we use this USB connection to get data at higher frequencies? Mavros over wifi seems to be capped at 10 hz with a lot of delay and message drop.

AndKe, 2020-10-23:
I tried to connect a few USB devices to the USB-OTG port of IMX6, but lsusb does not list them.. any ideas ?