3drobotics/solodevguide #331: Module mxc_v4l2_capture not found

tasarmiento, 2017-02-27:
Using libsolocam.so compiled for my system (PC running Ubuntu 14.04), or running modprobe results in the error:

modprobe: FATAL: Module mxc_v4l2_capture

I would like to create this module and insert it into the kernel, but can someone tell me where it resides on their Ubuntu 14.04 install?

kevinmehall, 2017-02-27:
You can only run libsolocam on Solo's onboard computer. `mxc_v4l2_capture` is the driver for Solo's hardware that receives the HDMI stream from the GoPro. This hardware doesn't exist on your PC, so that driver won't work there. You could perhaps adapt libsolocam to use a webcam via V4L2, but there are probably better video capture library options that already exist for desktop Linux.

closed 2017-02-27 by kevinmehall

tasarmiento, 2017-02-27:
Thanks @kevinmehall 
I came to this realization about the same time you responded.
I really just want to use the live stream in OpenCV on my laptop for troubleshooting before deploying to Solo and the sololink.sdp gives me empty frames in OpenCV. 
I will investigate some other options.